TIS2020-FMFI/krokovy-motor #11: Komunikácia s krokovým motorom

blacktea1111994, 2020-10-02:
Možnosti ovládania čipu PICAXE v Java knižnici.

pavelpetrovic, 2020-10-03:
Ten čip by mal byť prepojený sériovým káblom tak isto ako bol prepojený tatrabot s PC - takým istým alebo veľmi podobným USB prevodníkom, ktorý na PC vytvorí virtuálny sériový port - napr. COM3.  Predpokladám, že ten čip bude programovať technik, takže s ním sa len dohodnite na komunikačnom protokole. Východiskom by mohlo byť napr:
znak '+' - posun o jeden krok v kladnom smere
znak '-' - posun o jeden krok v zápornom smere
ešte by bolo možno pekné, keby ten čip do sériového portu pravidelne (napr. raz za sekundu) vysielal nejaký znak aby sa:
1) dalo automaticky detekovať v akom porte je pripojený a nemusel to používateľ nastavovať ručne - ako sme museli napr. pri tatrabotoch v programe putty
2) bola priebežná kontrola spojenia, že to beží a je pripojené (zároveň môžete používateľa upozorniť, keď sa čip vypne - a teda prestane vysielať - môže to byť povedzme vizualizované ako nejaký stav ON/OFF).

Java ma ten problemik, ze nema standardne API na komunikaciu cez seriovy port...
Dlho sa pouzivala proprietarna kniznica RXTX, ale casom prestala byt udrziavana a vraj uz s JDK8 nesla skompilovat bez uprav.
Vraj sa da pouzit http://javadox.com/org.scream3r/jssc/2.8.0/javadoc/jssc/package-summary.html

Pred dvomi rokmi v podobnej situacii pouzili studenti toto:
https://fazecast.github.io/jSerialComm/
mozete sa pozriet ako to pouzili tu:
https://github.com/TIS2018-FMFI/urychlovac/blob/master/src/main/java/VacControllerCommunication.java

Bolo by dobre to otestovat na niecom inom predtym ako budete testovat na tom motore.
ak niekto z Vas ma arduino - tak to je idealny sposob ako to otestovat ak  nie, tak sa bud niekedy volakto zastavte osobne alebo mi poslite mailom adresu a poslem arduino postou.

blacktea1111994, 2020-10-03:
@pavelpetrovic 
Ja mám Arduino-set od vás z DTV. Takže budeme vedieť to otestovať pomocou neho.

blacktea1111994, 2020-10-24:
v priečinku jSerialCommunication  sa nachádza:
- README.txt s popisom testovacieho programu
- program pre Arduino imitujúci komunikáciu s čipom krokového  motora

blacktea1111994, 2020-10-24:
Do priečinku jSerialCommunication vo vetve Documentation som pripojil jar file pre kniznicu jserialcomm a vetvu uzatvaram, pretoze obsahuje vsetko potrebne a splnila ucel - testovania seriovej komunikacie pomocou arduina.

closed 2020-10-26 by FilipSekerka