TIS2020-FMFI/krokovy-motor #15: Vypracovanie špecifikácie požiadaviek

MatejKomlossy, 2020-10-09:


pavelpetrovic, 2020-10-19:
Dobry den. Prestudoval som si poslednu verziu KP a dopisal som do nej moje komentare. Prosim zapracujte ich podla moznosti a potom dajte este vyslednu verziu KP okomentovat a potom schvalit zadavatelovi.

pavelpetrovic, 2020-10-19:
priloha...
[katalogPoziadaviekEdit_s_komentarmi.docx](https://github.com/TIS2020-FMFI/krokovy-motor/files/5401644/katalogPoziadaviekEdit_s_komentarmi.docx)

FilipSekerka, 2020-10-19:
Zapracovali sme vase pripomienky, posielame opravenu verziu
 https://docs.google.com/document/d/1_8cBW3x6xfxcjxEZgmq_nBVBfYQ1V87Vm6R6Uz2ew-w/edit
@pavelpetrovic

pavelpetrovic, 2020-10-20:
skvele. do dokumentu som pridal dalsie moje drobnosti a komentare, skuste prosim spracovat.

FilipSekerka, 2020-10-20:
@pavelpetrovic zapracovali sme pripomienky
https://docs.google.com/document/d/1_8cBW3x6xfxcjxEZgmq_nBVBfYQ1V87Vm6R6Uz2ew-w/edit

pavelpetrovic, 2020-10-20:
dakujem, dobra praca. docistite teda prosim vsetky komentare (pridal som tam uz len start,end v tej kalibracii, aby som tomu uz celkom rozumel) a poslite to zadavatelovi, aby to schvalil - potrebujeme jeho pisomny (mailovy) suhlas, ze tato verzia uz obsahuje vsetky jeho poziadavky.

FilipSekerka, 2020-10-22:
@pavelpetrovic 
Prosim, pisal som p.Gregusovi kvoli posielaniu toho kontrolneho znaku '@' (ci je cip zapnuty a pripojeny).
Odpisal mi, ze ci nestaci ten znak '@' vysielat len po zapnuti cipu a potom neskor pocas merani uz nie.

Mam mu teda odpisat, ze urcite potrebujeme neustale posielanie ? 
Alebo staci iba na zaciatku, ako navrhol on ?

Dakujem

pavelpetrovic, 2020-10-22:
Zdá sa mi to prijateľný kompromis, po prijatí znaku '#' by mohol prestať vysielať tie '@'. 
Najmä, ak by okrem znakov + a - ešte prijal aj napr. znak '!', na ktorý by zasa odpovedal jedným ďalším '@', ale nevysielal by už tie '@' sám od seba. Čo by povedal na toto?

FilipSekerka, 2020-10-24:
@pavelpetrovic 
p.Gregus povedal, ze ten komunikacny protokol bude konzultovat v utorok priamo s technikom, ktory bude programovat ten cip a potom nam da vediet.

Prosim, dalej sa pyta na nastavenia seriovej komunikacie.
Pyta sa, ci by mohla byt rychlost prenosu 9600 a ci treba specialne nastavit
dalsie parametre(parita, pocet stop bitov alebo pripadne nejake dalsie), alebo tie ostatne parametre nie su potrebne.

Dakujem

pavelpetrovic, 2020-10-26:
jasne, ved to mu viete aj sami povedat.
klasicke 9600,8,N,1 bez flow control je standard.

FilipSekerka, 2020-11-09:
[image]

potvrdenie od zadavatela

closed 2020-11-09 by FilipSekerka