TIS2020-FMFI/krokovy-motor #26: Vytvorenie UML component diagram

FilipSekerka, 2020-10-23:


FilipSekerka, 2020-10-26:
@pavelpetrovic 
Spravili sme prvu verziu komponentneho diagramu, mohli by ste to, prosim, pozriet ?

https://drive.google.com/file/d/147MzQwAFvDN9MkF9_64OeW98fyyKnp-3/view?usp=sharing

Dakujeme

pavelpetrovic, 2020-10-26:
* dakujem, vyzera to vcelku optimisticky.
* ak by ste to chceli vylepsit, mozete tie interfejsy aj pomenovat.
* medzi tym picaxe a picaxe communication staci mat jednu gulicku na strane picaxe - je to iba jedno rozhranie. ze rozhranie mozu data prudit oboma smermi (aspon tuto jednu vec by som urcite upravil)
* zda sa mi, ze z komponentu measurement handler by sa este dalo oddelit nieco, co vie exportovat meranie do suborov 
* v tomto kontexte mi este napadlo, ze by bolo skvele, keby si aplikacia medzi jednotlivymi spustaniami dokazala zapamatat, do ktoreho priecinku na disku to pouzivatel naposledy exportoval - napriklad v nejakom tajnom konfigu :-), pripadne aj posledne nastavenia spektrometra => to by mohol byt potom samostatny komponentik config.

FilipSekerka, 2020-10-30:
https://drive.google.com/drive/folders/1hywsR6p_vKrIQi3_nSwrx5BUelBsW2xx

MatejKomlossy, 2020-11-01:
finalna verzia https://drive.google.com/file/d/1Nl2Av-YuapoMhF82lriPS7xPM1x7dFio/view?usp=sharing

closed 2020-11-01 by MatejKomlossy